thiagoralves/OpenPLC_Editor #9: Needs python update

pgiffuni, 2020-01-22:
From the install.sh, it seems like this tries to install python 2.7, which was (finally) deprecated this year, and is being removed from most linux distributions.

thiagoralves, 2020-01-22:
I'm very aware of this, however, OpenPLC Editor being a fork from Beremiz, which is a fork from PLCOpen Editor, relies on ancient libraries and tools. Having it up to date with Python 3 means basically to rewrite the whole thing (at least the wxwidgets-GUI portion). Unfortunately I just don't have the willpower to do it now, so people will have to install deprecated packages if they want to run OpenPLC Editor

pgiffuni, 2020-01-24:
Hmm.. I guess we can close the issue then :(

closed 2020-01-24 by pgiffuni