thiagoralves/OpenPLC_Editor #10: Add feature to simulate the automation plant (external python script model)

Matheus-Garbelini, 2020-03-10:
Hi @thiagoralves , I've made changes to the editor which adds a useful feature to simulate the softPLC runtime generated by the editor in a more realistic setup before deploying to the real OpenPLC v3 runtime. However, the ticks rate are kept to real-time (1000000) and incremented based on the minimum scan cycle configure on the editor (res0)

The changes are available in this commit: https://github.com/Matheus-Garbelini/OpenPLC_Editor/commit/8e091671b8799582caa909e333e572aa7c972709

In short, once a simulation starts, an external python script (model) is executed every scan cycle. The user-supplied external script can then read all PLC outputs and modify the PLC inputs according to an automation model of choice. The simulation running an [example script](https://github.com/Matheus-Garbelini/OpenPLC_Editor/blob/master/example_project/model.py) is shown in the Figure below: 
[image]

It's also possible to run the simulation in a standalone setup, i.e., you can start the model with the PLC logic without using the IDE. This is useful if the user only wants to test the model with a realistic PLC logic. In this case, it's necessary the addition of a [helper class](https://github.com/Matheus-Garbelini/OpenPLC_Editor/blob/master/example_project/plc_runner.py) that initializes the PLCObject.py, thus executing the user-supplied model:

[image]

On the user side. The external script can be really simple and mimics the Arduino convention by using setup and loop:
```Python
# Setup is called before the PLC program starts
def setup(PLC, args):
    # Initialize sample rate for a standalone model
    Stage1.Sample_rate = 60.0
    # Sets the Initial water level on PLC.
    PLC.LIT101.value = Stage1.LIT101

# model is called each PLC scan time (configured on IDE)
# Receive all variables from PLC program and return the altered variables
def loop(PLC, args):
    # Read PLC outputs
    MV101 = PLC.MV101.value
    P101 =  PLC.P101.value
    # Execute process model (stage 1)
    LIT101 = Stage1.process(MV101, P101)
    # Update PLC input
    PLC.LIT101.value = LIT101
```

**The question**
As of now, the simulation time is fixed to the PLC scan cycle, this means that if the user wants to speed up the simulation (run faster than real-time), he needs to manually reduce the PLC scan cycle and adjust timing behavior of the model. It would be useful if the PLCController.py could somewhat increase the rate on which the PLC ticks are incremented on the softPLC so the simulation could accelerate. Is there a viable way to change the PLCcontroller.py generated .c code to change the tick rate based on user setting? As I understand, the current reference for the ticks are 1000000 (1ms)

Thanks a lot for this editor and best regards.

thiagoralves, 2020-05-08:
This is great Matheus. Can you make this as a pull request? I'm pretty sure a lot of users will benefit from it. Thanks!

Matheus-Garbelini, 2020-05-21:
Hi @thiagoralves. I will update you by next week. Pause feature was implemented for windows only, Linux is pending this feature. Also, I'll need to remove some pictures I've modified before submitting the pull (logo has changed).

Regards.