thiagoralves/OpenPLC_Editor #11: Install compile errors, application won't successfully start

rdingham, 2020-04-09:
when running install.sh, on Raspberry Pi 4, I get a lot of compiler warnings like:

<code>
array_range_check.cc:202:36: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
    STAGE3_ERROR(0, symbol, symbol, "Subrange has lower limit (%"PRId64") larger than upper limit (%"PRId64").", GET_CVALUE( int64, symbol->lower_limit), GET_CVALUE( int64, symbol->upper_limit));
                                    ^
array_range_check.cc:202:71: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
    STAGE3_ERROR(0, symbol, symbol, "Subrange has lower limit (%"PRId64") larger than upper limit (%"PRId64").", GET_CVALUE( int64, symbol->lower_limit), GET_CVALUE( int64, symbol->upper_limit));
                                                                       ^
array_range_check.cc:212:36: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
    STAGE3_ERROR(0, symbol, symbol, "Subrange has lower limit (%"PRIu64") larger than upper limit (%"PRIu64").", GET_CVALUE(uint64, symbol->lower_limit), GET_CVALUE(uint64, symbol->upper_limit));
                                    ^
array_range_check.cc:212:71: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
    STAGE3_ERROR(0, symbol, symbol, "Subrange has lower limit (%"PRIu64") larger than upper limit (%"PRIu64").", GET_CVALUE(uint64, symbol->lower_limit), GET_CVALUE(uint64, symbol->upper_limit));
                                                                       ^
ar: `u' modifier ignored since `D' is the default (see `U')
Making all in stage4
Making all in generate_c
In file included from generate_c.cc:525:
generate_c_il.cc: In member function virtual void* generate_c_il_c::visit(il_function_call_c*):
generate_c_il.cc:957:11: warning: this if clause does not guard... [-Wmisleading-indentation]
           if (nb_param > 0)
           ^~
generate_c_il.cc:959:13: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the if
             if (param_value == NULL) {
             ^~
ar: `u' modifier ignored since `D' is the default (see `U')
Making all in generate_iec
ar: `u' modifier ignored since `D' is the default (see `U')
config.status: creating config/config.h
config.status: config/config.h is unchanged
</code>

When I try to run OpenPLC Editor I get:

<code>
No module named 'lxml'
</code>

thiagoralves, 2020-04-09:
That’s because the editor was never meant to be run on a Pi. It should run on your desktop. Now, if you can figure out a way to install lxml on the Pi then it might work

jaketho, 2020-05-08:
It looks like you should be able to install lxml on the pi, not sure if this will enable you to run the editor though. [Stack Exchange Link](https://raspberrypi.stackexchange.com/questions/68894/cant-install-lxml)

`sudo apt-get install python3-lxml python-lxml`

`sudo apt-get install libxml2-dev libxslt-dev python-dev`

`pip install lxml --user`