thiagoralves/OpenPLC_Editor #19: Docker image to handle the old Python

dcihlar, 2020-11-11:
On the newer systems the old Python 2 is now completely gone, so installing the current OpenPLC Editor is a hassle.

I propose a Docker image as a workaround until the editor is ported to Python 3:

```Dockerfile
FROM debian:buster

RUN apt-get update && \
    apt-get install -y \
        sudo curl python-pip apt-utils coreutils vim-gtk dbus-x11 bash-completion \
        build-essential pkg-config bison flex autoconf automake libtool make git \
        libssl-dev python2 python-wxgtk3.0 python2-dev libpng-dev libfreetype6-dev

# password = user
RUN mkdir -p /etc/skel/.local/share/applications && \
    mkdir -p /root/.local/share/applications && \
    useradd user --uid 1000 -p aaBrLCcg4bKmQ --groups sudo --create-home

RUN git clone --depth 1 --single-branch https://github.com/thiagoralves/OpenPLC_Editor.git && \
    cd OpenPLC_Editor && \
    ./install.sh && \
    mv openplc_editor.sh /usr/local/bin/openplc-editor

USER user
ENV HOME=/home/user
ENV PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/home/user/.local/bin
ENV LANG=en_US.utf8
WORKDIR /home/user

CMD openplc-editor
```

Build it with:
```bash
docker build -t openplc .
```

And run with:
```bash
#!/bin/bash
mkdir -p data # this is where projects can be saved
docker run \
	--env="DISPLAY" \
	--volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" \
	--volume="$(pwd)/data:/home/user/data" \
	openplc
```

ALTracer, 2021-09-10:
@dcihlar, can you please rewrite the Dockerfile to use multi-stage builds?

Target image should not have to carry build dependencies (autotools, git, python2-dev, libfreetype6-dev, libpng-dev) once the builder image completed installing the runtime dependencies.
While `FROM debian:buster` is 110MB, the very next RUN adds 880MB of packages (without issuing `apt-get clean`), and RUN install.sh adds some 360MB more. The resulting build is 1.36 GB for a single tool (Beremiz/wxpython+matiec+gcc)

My Docker-fu is not strong enough :smile:

vinimlo, 2022-10-25:
Works very well!! I would love to see this dockerfile on the main branch!