thiagoralves/OpenPLC_Editor #23: Can't install on debian 11

Sirman13, 2021-11-12:
E: Impossibile trovare il pacchetto python-wxgtk3.0
E: Impossibile trovare alcun pacchetto tramite il glob "python-wxgtk3.0"
E: Impossibile trovare alcun pacchetto tramite l'espressione regolare "python-wxgtk3.0"

thiagoralves, 2021-11-12:
Debian 11 removed support for python-wxgtk3.0. It now only has wxgtk4.0 and that won't work with OpenPLC Editor. You can try to install wxgtk3.0 from debian 10 using a method similar to this: https://openplc.discussion.community/post/open-plc-editor-9298979?pid=1300091863 or you can just use Debian 10

datamaskinen, 2021-11-13:
make an appimage of this when using packages not in main repositorys....

https://appimage.org/

thiagoralves, 2022-01-20:
This issue was fixed by including wxgtk3.0 on the repository

closed 2022-01-20 by thiagoralves