thiagoralves/OpenPLC_Editor #37: Importing LD without connection collection results in incorrect auto placement

pekka-28, 2022-05-07:
When importing a single rung network without connection elements for the connector routing, the auto routing connects to the connectionPointIn of the incoming element to the connectionPointIn of the outgoing element instead of the connectionPointOut.

To reproduce, create a single rung network with a contact and coil and save. Edit PLC.xml by removing sub-elements in connection collection. Open project again. 

Also, if the positions of the elements is removed, only the LeftPower rail is drawn and errors are reported if additional elements are added to the network.

thiagoralves, 2022-05-12:
The editor really shouldn't be able to process corrupted or incomplete project files. It appears to me that this scenario you described is impossible to occur without externally editing the .xml project file.

closed 2022-05-12 by thiagoralves

pekka-28, 2022-05-12:
The IEC schema indicates the elements are optional. Our use case is to generate ladder logic code from a higher level representation and we would like to avoid the mechanical transcription into the application.

I can understand your reluctance to support complete auto-layout, but as the file is read in, it just looks like the incorrect port is selected when no connection path is provided. It makes our use of the application difficult as we don’t wish to implement a connection layout router in our tool as it does not consider the geometry of the code presentation at all.

I did look through the code in the editors to see where the incorrect port was selected, but could not locate where the code is read and the link established. If you can point me to the source that creates the reference, I would probably be able to provide a diff to correct this problem and improve robustness.

Regards
Pekka Pihlajasaari
--
***@***.***           Data Abstraction (Pty) Ltd         +27 11 484 9664