thiagoralves/OpenPLC_Editor #53: Check for updates (Windows) failing

JohnnyOpcode, 2022-08-04:
The current version downloaded from the website detects a new version, but fails to download (?) and install (?) in Windows (10).

ANy ideas where to patch?

thiagoralves, 2022-08-04:
It is working for me. What is the error message you're getting?

JohnnyOpcode, 2022-08-05:
It hangs shortly after the download is my best interpretation.
[image]

thiagoralves, 2022-08-05:
Check your internet connection. There could be something blocking access to
github servers. Make sure there aren't any firewall or antivirus blocking
OpenPLC Editor traffic.

JohnnyOpcode, 2022-08-05:
Do you mean I need to add a firewall rule (manually) for the application under Windows 10? I am browsing GitHub without issue as can be seen here.

thiagoralves, 2022-08-05:
No add a firewall rule, but make sure there isn't any rule *blocking* it.
Firewall rules are per application, so your browser could browse github
normally, but it could be possible that there is a rule blocking OpenPLC
Editor external communications

JohnnyOpcode, 2022-08-09:
C:\Users\DEV\OpenPLC_Editor>if exist .\new_editor\ rmdir /s /Q new_editor

C:\Users\DEV\OpenPLC_Editor>if exist .\OpenPLC_Editor\ rmdir /s /Q OpenPLC_Editor

C:\Users\DEV\OpenPLC_Editor>git clone https://github.com/thiagoralves/OpenPLC_Editor
Cloning into 'OpenPLC_Editor'...
fatal: unable to access 'https://github.com/thiagoralves/OpenPLC_Editor/': error setting certificate verify locations:  CAfile: C:\Program Files\Git\mingw64\ssl\certs\ca-bundle.crt CApath: none

C:\Users\DEV\OpenPLC_Editor>if exist .\OpenPLC_Editor\editor\ (
move .\OpenPLC_Editor\editor .\new_editor
 copy /y .\OpenPLC_Editor\revision .\
 rmdir /s /Q OpenPLC_Editor
 echo "Update applied successfully"
)  else (echo "Error cloning from repository!" )
"Error cloning from repository!"

C:\Users\DEV\OpenPLC_Editor>

thiagoralves, 2022-08-09:
The updater is trying to use Git from your machine but apparently it is
broken (missing certificates). OpenPLC Editor includes its own Git. Try
reinstalling Git on your machine, or removing it altogether, so that the
updater will use the bundled Git.

JohnnyOpcode, 2022-08-09:
I have no Git installed or on my path. I do have GitHub Desktop however.

This directory does not exist on my PC nor the .crt file as well.
C:\Program Files\Git\mingw64\ssl\certs\ca-bundle.crt

I will keep debugging the problem..

LeaderXinjianSun, 2025-12-14:
Hi, Can I change defult url 'https://github.com/thiagoralves/OpenPLC_Editor.git' to my forked Repository 'https://gitee.com/sunxinjian/OpenPLC_Editor.git'? It may be faster than clone from github directly.