thiagoralves/OpenPLC_Editor #55: Can't view changes while working on adding a new feature without compiling

MahmoudDawood, 2022-08-29:
I'm currently working on adding a new feature to OpenPLC.
The problem is when I edit the source code I can't test my changes and view them without reinstalling the editor all over again.
Is there a quick way for developers to view changes before compiling?

thiagoralves, 2022-08-29:
You're definitely doing something wrong. The editor was written in Python, any change on the code will show up once you close and open the editor again, unless there is some really weird configuration on your setup that is forcing Python to use the cached .pyc files instead of your modified .py source code. In that case, just deleting all .pyc files should fix it.

closed 2022-08-29 by thiagoralves

MahmoudDawood, 2022-08-29:
@thiagoralves 
That's not what I meant, the Python files keep the changes as expected.
I'm concerned about testing the changes I made and how to view them in a GUI without the need to reinstall OpenPLC  all over again to see the changes applied
Is there a way to do so?

thiagoralves, 2022-08-29:
There is nothing else but Python code on the editor (except the runtime that is written in C). Even the GUI elements were written in Python. So, as I said, any changes to the Python code must reflect immediately on both appearance and functionality of the OpenPLC Editor. In fact, reinstalling OpenPLC Editor will revert back to the original code and all your changes will be lost, unless you're doing something very wrong.

MahmoudDawood, 2022-08-29:
@thiagoralves 
Ok, I've been using windows, changing the source code, and trying to install with changes applied, but then tried modifying the installed files directly and it actually **showed up** in the Editor!
#### Thanks, Thiago!