thiagoralves/OpenPLC_Editor #56: Couple small issues with python3... and fixes

nolange, 2022-10-08:
Should be pretty self explanatory, those warn or raise exceptions on python3.10 + python3-wxgtk4.0.

Thought about setting up a PLC Tool on a ChromeBook for a relative, did not expect this kinda troubles.

Doesn't help that the links to github are rather hidden on the website.

```patch
diff --git a/editor/BeremizIDE.py b/editor/BeremizIDE.py
index c053adc..e7fa1fe 100644
--- a/editor/BeremizIDE.py
+++ b/editor/BeremizIDE.py
@@ -122,7 +122,7 @@ class Beremiz(IDEFrame):
                    kind=wx.ITEM_NORMAL, text=_(u'Open') + '\tCTRL+O')
         parent.AppendSubMenu(self.RecentProjectsMenu, _("&Recent Projects"))
         parent.AppendSeparator()
-        parent.AppendMenu(wx.ID_ANY, _("&Tutorials and Examples"), self.TutorialsProjectsMenu)
+        parent.Append(wx.ID_ANY, _("&Tutorials and Examples"), self.TutorialsProjectsMenu)
 
         examples_dir = Bpath("examples")
         project_list = sorted(os.listdir(examples_dir))
diff --git a/editor/Beremiz_service.py b/editor/Beremiz_service.py
index 475c2b6..5d45748 100755
--- a/editor/Beremiz_service.py
+++ b/editor/Beremiz_service.py
@@ -341,7 +341,7 @@ if enablewx:
                 _servicename = self.pyroserver.servicename
                 _servicename = '' if _servicename is None else _servicename
                 dlg = ParamsEntryDialog(None, _("Enter a name "), defaultValue=_servicename)
-                dlg.SetTests([(lambda name: len(name) is not 0, _("Name must not be null!"))])
+                dlg.SetTests([(lambda name: len(name) != 0, _("Name must not be null!"))])
                 if dlg.ShowModal() == wx.ID_OK:
                     self.pyroserver.servicename = dlg.GetValue()
                     self.pyroserver.Restart()
diff --git a/editor/controls/ProjectPropertiesPanel.py b/editor/controls/ProjectPropertiesPanel.py
index bf10fe6..a14f811 100644
--- a/editor/controls/ProjectPropertiesPanel.py
+++ b/editor/controls/ProjectPropertiesPanel.py
@@ -256,8 +256,8 @@ class ProjectPropertiesPanel(wx.Notebook):
             elif item == "scaling":
                 for language, (x, y) in value.items():
                     if language in self.Scalings:
-                        self.Scalings[language][0].SetValue(x)
-                        self.Scalings[language][1].SetValue(y)
+                        self.Scalings[language][0].SetValue(int(x))
+                        self.Scalings[language][1].SetValue(int(y))
             else:
                 tc = getattr(self, item, None)
                 if tc is not None:
```

thiagoralves, 2022-10-10:
Thanks for the fixes. Could you make a PR with those please?

PS: You should expect all kind of trouble when using a dev branch. If you want a smoother experience, try using the website installer (master branch) on a supported platform (Debian and Fedora variants).

nolange, 2022-10-10:
The master branch is quite impossible to get to work on debian bookworm. There's no python2 and using pip yielded some broken wxpython libraries.

Sorry, don't think those 3 changes are worth doing a branch & PR. I am not following this further