thiagoralves/OpenPLC_Editor #60: builder.py can't handle non-ASCII characters in logs

jpvg10, 2022-11-20:
I was trying to upload a program into a microcontroller and ran into this error:

[image]

Some directories in my system have non-ASCII characters in them, since my computer is in Spanish. I also see some Spanish text in the logs. I managed to make it work by doing the following change in `editor/arduino/builder.py` line 410.

```diff
-compiler_logs += stdout
+compiler_logs += repr(stdout)
```

Then I restarted the editor and tried to upload again, and it worked.

I don't know if this is the best solution, and whether these is the only place in the code where this is a problem (probably not). It would be worth looking into it.

thiagoralves, 2022-11-20:
This bug has already been fixed ages ago. See commit [4884cef6d8de2168346128032cf90abc06711a5b](https://github.com/thiagoralves/OpenPLC_Editor/commit/4884cef6d8de2168346128032cf90abc06711a5b)

Please make sure you are running the latest version of the OpenPLC Editor

closed 2022-11-20 by thiagoralves

jpvg10, 2022-11-20:
Thanks for the quick reply. Indeed I updated and it now works without issues!
I downloaded the Windows installer from the website 2 weeks ago, I guess the latest version wasn't there then.