thiagoralves/OpenPLC_Editor #76: Cannot install on macOS Ventura 13.x

mbollini, 2023-05-03:
Installation complete without errors but starting OpenPLC Editor I get:

_This will install OpenPLC Editor dependencies in your system:
  - Brew
  - Python3
  - wxPython

Would you like to continue?_

mbollini, 2023-05-03:
It goes in a loop. Every time I launch OpenPLC Editor I got the same dialog box, I answer Yes, terminal window appears, requires my sudo password, do some installations and terminate. As I relaunch OpenPLC Editor this loop restart.

thiagoralves, 2023-05-03:
Any error messages during the installation process? Can you copy/paste here the logs from the terminal window?

mbollini, 2023-05-03:
_Installing Xcode Command Line Tools...
xcode-select: error: command line tools are already installed, use "Software Update" in System Settings to install updates
Installing Brew...
==> Checking for `sudo` access (which may request your password)...
Password:
==> This script will install:
/usr/local/bin/brew
/usr/local/share/doc/homebrew
/usr/local/share/man/man1/brew.1
/usr/local/share/zsh/site-functions/_brew
/usr/local/etc/bash_completion.d/brew
/usr/local/Homebrew

Press RETURN/ENTER to continue or any other key to abort:
==> /usr/bin/sudo /usr/sbin/chown -R maurizio:admin /usr/local/Homebrew
==> Downloading and installing Homebrew...
HEAD is now at fec24f5c0 Merge pull request #15352 from woodruffw-forks/ww/step-summary-var
==> Installation successful!

==> Homebrew has enabled anonymous aggregate formulae and cask analytics.
Read the analytics documentation (and how to opt-out) here:
  https://docs.brew.sh/Analytics
No analytics data has been sent yet (nor will any be during this install run).

==> Homebrew is run entirely by unpaid volunteers. Please consider donating:
  https://github.com/Homebrew/brew#donations

==> Next steps:
- Run these two commands in your terminal to add Homebrew to your PATH:
    (echo; echo 'eval "$(/usr/local/bin/brew shellenv)"') >> /Users/maurizio/.zprofile
    eval "$(/usr/local/bin/brew shellenv)"
- Run brew help to get started
- Further documentation:
    https://docs.brew.sh
============================
All dependencies were successfully installed. You can now close this terminal window
============================_

Launch OpenPLC Editor, I get this message window:
_OpenPLC Editor dependencies were not installed correctly. Please relaunch the app to install them again._

I press "Ok"
Launch OpenPLC Editor and I get:
_This will install OpenPLC Editor dependencies in your system:
  - Brew
  - Python3
  - wxPython

Would you like to continue?_

thiagoralves, 2023-07-24:
The installer was updated and possibly fixes this. Could you try again please?

closed 2023-08-07 by thiagoralves

mbollini, 2023-08-07:
Tested, same issue.

madmax76, 2024-01-23:
Still a problem in 2024, it has python dependency problems that I am unable to resolve on 14.3 Sonoma.

==> matplotlib.backends.backend_wxagg

thiagoralves, 2024-01-23:
Have you downloaded the precompiled version from Autonomy website? Getting it straight from sources can be challenging. The error you’re facing is related to a preinstalled matplotlib that doesn’t match the version required by OpenPLC Editor. If you’re not a developer and just want to use the editor, the easiest path is to just download and install the prebuilt version from Autonomy