thiagoralves/OpenPLC_Editor #87: Compile Error

eilarraza, 2023-08-06:
Hello All,

Running Description:	Ubuntu 20.04.6 LTS
OpenPLC Editor 2.01 Release On 2023-08-03

I am a newbie trying OpenPLC for the first time.

While trying to **Generate a Program for OpenPLC runtime** or **transfer program to PLC** (in my case barebone arduino nano) you got an error related to the newly added file **jaguar.txt** file.

Iniciar la construcción en /home/user/folder/PLC-Examples/PLC-Example-NanoBoard/build
Generando el código ST/IL/SFC IEC-61131  para el SoftPLC...
    Collecting data types
    Collecting POUs
    Generate POU PLC_Example_Nano
    Generate Config(s)
Compilando el programa IEC en código C...
"/home/user/Documentos/Diario/Referencia/Software/OpenPLC/OpenPLC_Editor/matiec/iec2c"  -f -l -p -I "/home/user/Documentos/Diario/Referencia/Software/OpenPLC/OpenPLC_Editor/matiec/lib" -T "/home/user/folder/Diario/2023/PLC-Examples/PLC-Example-NanoBoard/build" "/home/user/PLC-Examples/PLC-Example-NanoBoard/build/plc.st"
Terminado con estatus 1 (pid 26548)
Error opening included file jaguar.txt

Error: El compilador de IEC a C retorno 1 

I got this error for _every provided example_ like blink or hello world. 

The jaguar.txt file is located here:

**/OpenPLC/OpenPLC_Editor/matiec/lib**

And here:

**/OpenPLC/OpenPLC_Editor/editor/arduino/src/lib**

May you please double check?

Thanks a lot.

Best Regards

eilarraza, 2023-08-07:
Hello, 

I just tested with the old OpenPLC version from https://openplcproject.com/download-linux/ and the compile / execution runs Ok. So the problem is with this new release:

**OpenPLC Editor 2.01 Release On 2023-08-03**

Best Regards

thiagoralves, 2023-08-07:
Fixed

closed 2023-08-07 by thiagoralves