thiagoralves/OpenPLC_Editor #91: Arduino Uno R3 | Error during build --> PWM_Controller

Irrlich2142, 2023-10-23:
Hey There,
I can't compile a testprogramm for the PWM_Controller.

Editor Running on Windows Computer
Downloaded the latest masterbranch
Target: Arduino Uno R3

Im just added an PWM_Controller to the FUP_Programm and added some Variables matching the Informations to the Block.
Here one Screenshot of it:
[image]

**When i want to load it onto my Arduino it gives out this Error:**
```
C:\Users\pepe_\AppData\Local\Temp\ccNnGypV.ltrans0.ltrans.o: In function `PWM_CONTROLLER_body__':
C:\Users\pepe_\OpenPLC_Editor/editor\arduino\src\lib/arduino_lib_FB.h:524: undefined reference to `set_hardware_pwm'
collect2.exe: error: ld returned 1 exit status

Error during build: exit status 1

COMPILATION FAILED!
```

**Here the full Compilation Log:**

```
Compiling .st file...
POUS.c
POUS.h
LOCATED_VARIABLES.h
VARIABLES.csv
Config0.c
Config0.h
Res0.c
Generating binary file...

COMPILATION START: arduino:avr:uno
Using board 'uno' from platform in folder: C:\Users\pepe_\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
Using core 'arduino' from platform in folder: C:\Users\pepe_\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
Detecting libraries used...
"C:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-Ieditor\\arduino\\src\\lib" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "C:\\Users\\pepe_\\AppData\\Local\\Temp\\arduino-sketch-F608021BE6D63A357997DDB0A5EF2358\\sketch\\Baremetal.ino.cpp" -o nul
Alternatives for Arduino_OpenPLC.h: [src]
ResolveLibrary(Arduino_OpenPLC.h)
  -> candidates: [src]
"C:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-Ieditor\\arduino\\src\\lib" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IC:\\Users\\pepe_\\OpenPLC_Editor\\editor\\arduino\\src" "C:\\Users\\pepe_\\AppData\\Local\\Temp\\arduino-sketch-F608021BE6D63A357997DDB0A5EF2358\\sketch\\Baremetal.ino.cpp" -o nul
"C:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-Ieditor\\arduino\\src\\lib" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IC:\\Users\\pepe_\\OpenPLC_Editor\\editor\\arduino\\src" "C:\\Users\\pepe_\\AppData\\Local\\Temp\\arduino-sketch-F608021BE6D63A357997DDB0A5EF2358\\sketch\\ModbusSlave.cpp" -o nul
"C:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-Ieditor\\arduino\\src\\lib" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IC:\\Users\\pepe_\\OpenPLC_Editor\\editor\\arduino\\src" "C:\\Users\\pepe_\\OpenPLC_Editor\\editor\\arduino\\src\\Config0.c" -o nul
"C:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-Ieditor\\arduino\\src\\lib" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IC:\\Users\\pepe_\\OpenPLC_Editor\\editor\\arduino\\src" "C:\\Users\\pepe_\\OpenPLC_Editor\\editor\\arduino\\src\\POUS.c" -o nul
"C:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-Ieditor\\arduino\\src\\lib" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IC:\\Users\\pepe_\\OpenPLC_Editor\\editor\\arduino\\src" "C:\\Users\\pepe_\\OpenPLC_Editor\\editor\\arduino\\src\\Res0.c" -o nul
"C:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-Ieditor\\arduino\\src\\lib" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IC:\\Users\\pepe_\\OpenPLC_Editor\\editor\\arduino\\src" "C:\\Users\\pepe_\\OpenPLC_Editor\\editor\\arduino\\src\\arduino.cpp" -o nul
"C:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-Ieditor\\arduino\\src\\lib" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IC:\\Users\\pepe_\\OpenPLC_Editor\\editor\\arduino\\src" "C:\\Users\\pepe_\\OpenPLC_Editor\\editor\\arduino\\src\\debug.c" -o nul
"C:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-Ieditor\\arduino\\src\\lib" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IC:\\Users\\pepe_\\OpenPLC_Editor\\editor\\arduino\\src" "C:\\Users\\pepe_\\OpenPLC_Editor\\editor\\arduino\\src\\glueVars.c" -o nul
Generating function prototypes...
"C:\\Users\\pepe_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-Ieditor\\arduino\\src\\lib" "-IC:\\Us [... cut ...] DB0A5EF2358" -lm
Using library src in folder: C:\Users\pepe_\OpenPLC_Editor\editor\arduino\src (legacy)

C:\Users\pepe_\AppData\Local\Temp\ccNnGypV.ltrans0.ltrans.o: In function `PWM_CONTROLLER_body__':
C:\Users\pepe_\OpenPLC_Editor/editor\arduino\src\lib/arduino_lib_FB.h:524: undefined reference to `set_hardware_pwm'
collect2.exe: error: ld returned 1 exit status

Error during build: exit status 1

COMPILATION FAILED!

```

Are there some missing files?

Thanks for the Reply.

Greetings Philipp

thiagoralves, 2023-10-23:
This means that the PWM_Controller block was not implemented for your
platform. In fact, this block has been implemented to only one platform so
far. Any other platform that needs support for this will have to implement
the set_hardware_pwm() function

Irrlich2142, 2023-10-24:
Do you mean platform (arduino uno) or the platform as windows with openplc? If its the arduino, do you know what arduino is the one who is implemented? 

Thanks for the reply

thiagoralves, 2023-10-24:
Platform as arduino uno. No arduino board has this implemented. Currently
only one prototype platform has code to support it. I can check to see what
the effort is to bring this to basic arduino unos (if that's even possible)

Irrlich2142, 2023-10-24:
That would be great. Can you give me feedback if you know if its possible or not?

thiagoralves, 2023-10-24:
Just added support for Arduino Uno: https://github.com/thiagoralves/OpenPLC_Editor/commit/bf469e7f2d24835e911c1ea09373fa8161e76129

The code compiles, but I haven't check functionality yet. Channel is mapped to the PWM output pins like %IW0, %IW1, etc. So, if you use channel 0 it will be the same as using %IW0 pin (which on Arduino Uno that's pin 9)

Irrlich2142, 2023-10-25:
Awsome, Thanks for the fast support. I'll try it when im back at home. I'll give you feedback afterwards. 

Best regards.

Irrlich2142, 2023-10-25:
As far a s i can tell:
Channel 0 --> Pin 9  Address works
Channel 1 --> Pin 10 Address works
Frequency works also. I tested 500Hz with 25% Duty and it is 2ms from peak to peak.
Duty cycle with 50% is correct and 25% is also correct.

Here a Picture of 500Hz with 25% duty cycle on my Oscilloscope:
[image]

Only Issues:
-Channel 2 is'nt working (i guess from the look at the code that this is intentional. Only pins 9 and 10 are defined)
-When i try to use 2 Blocks e.g Channel 0 and 1 only one works. Here the program:
[image]

Over all thank you very much for Implement this feature so quickly.

Best regards Philipp

Irrlich2142, 2023-10-25:
> Just added support for Arduino Uno: [bf469e7](https://github.com/thiagoralves/OpenPLC_Editor/commit/bf469e7f2d24835e911c1ea09373fa8161e76129)
> 
> The code compiles, but I haven't check functionality yet. Channel is mapped to the PWM output pins like %IW0, %IW1, etc. So, if you use channel 0 it will be the same as using %IW0 pin (which on Arduino Uno that's pin 9)

shouldn't it be %QW0 and %QW1 ...

thiagoralves, 2023-10-26:
> shouldn't it be %QW0 and %QW1 ...

Yes! My mistake…

thiagoralves, 2023-10-26:
> As far a s i can tell: Channel 0 --> Pin 9 Address works Channel 1 --> Pin 10 Address works Frequency works also. I tested 500Hz with 25% Duty and it is 2ms from peak to peak. Duty cycle with 50% is [...]

Pins 9 and 10 share the same timer (Timer1) on ATmega328, which means that they can only run if both are set for the same frequency. I haven’t done a deep dive into the atmega datasheet so I could be wrong on that one, but that’s normally how it works on most microcontrollers. This is probably why it is not working when you try to use two blocks.

About pin 11, I left it out because I thought it was using Timer0 which is reserved for millis() in the Arduino world. But I was wrong, it uses Timer2, so it is safe to bring it in as well

Irrlich2142, 2023-10-26:
Ive also noticed that i cant go under 250Hz. if i set the variable for the Frequency to e.g. 20.0 or 2.0 the output is still 250Hz.

thiagoralves, 2023-10-26:
That probably has to do with a limitation from the ATmega timer. Im using [this library](https://github.com/khoih-prog/AVR_PWM) to handle all the PWM stuff. This guy wrote the same lib for various architectures, which makes it easy to port over the same code to esp32, stm32, etc...

By the way, my latest commit added pin 11 to arduino uno boards. The problem with pin 11 was for the Leonardo (32u4). I made sure to remove that pin if the board is a Leonardo

closed 2023-11-07 by thiagoralves