thiagoralves/OpenPLC_Editor #99: ImportError: libjpeg.so.9: cannot open shared object file: No such file or directory

wrcrooks, 2024-01-25:
After running install script, I get the following error:

```
# ./openplc_editor.sh
Traceback (most recent call last):
  File "/home/user/Downloads/OpenPLC_Editor/./editor/Beremiz.py", line 32, in <module>
    import wx
  File "/home/user/Downloads/OpenPLC_Editor/.venv/lib/python3.10/site-packages/wx/__init__.py", line 17, in <module>
    from wx.core import *
  File "/home/user/Downloads/OpenPLC_Editor/.venv/lib/python3.10/site-packages/wx/core.py", line 12, in <module>
    from ._core import *
ImportError: libjpeg.so.9: cannot open shared object file: No such file or directory

```

Install Procedure:
git clone https://github.com/thiagoralves/OpenPLC_Editor
cd OpenPLC_Editor
sudo apt install gcc g++ bison flex autoconf automake make python3 python3-pip 
pip3 install wxPython lxml future matplotlib zeroconf pyserial
cd matiec
autoreconf -i
./configure
make -s
cp ./iec2c ../editor/arduino/bin/
cd ../
sudo chmod +x install.sh
./install.sh # Get this error:
```
E: Package 'python3.8' has no installation candidate
./install.sh: line 33: python3.8: command not found
./install.sh: line 34: /home/user/Downloads/OpenPLC_Editor/.venv/bin/python: No such file or directory
./install.sh: line 35: /home/user/Downloads/OpenPLC_Editor/.venv/bin/python: No such file or directory
./install.sh: line 36: /home/user/Downloads/OpenPLC_Editor/.venv/bin/python: No such file or directory
```
sudo nano install.sh # Replace all instances of "python3.8" with "python3" in the install.sh script
./install.sh # No error messages this time
./openplc_editor.sh # Returns the error about libjpeg.so.9

I checked and libjpeg-dev is installed on my system. But also ran `sudo apt install libjpeg-dev` just in case. Any ideas on this?

**System Info**
Operating System: Debian GNU/Linux 11
KDE Plasma Version: 5.20.5
KDE Frameworks Version: 5.78.0
Qt Version: 5.15.2
Kernel Version: 5.10.0-27-amd64
OS Type: 64-bit

thiagoralves, 2024-01-25:
Installer should take care of all dependencies. You don't have to install wxPython, lxml, etc on your system python as it is not being used. OpenPLC creates a virtual environment using venv for its own dependencies. Most likely this libjpeg.so.9 issue is a wxPython dependency that didn't get sorted out during the build process. Try changing python3.8 with python3.9 on the installer script. I know that wxPython has problems building on python 3.10 and up, that's why I included python3.8 on the script, but apparently your system does not offer support for python 3.8 anymore.

thiagoralves, 2024-01-25:
I just updated the install script to use python 3.9. Clone the repo and try again.

wrcrooks, 2024-01-25:
Deleted the downloaded repo folder, cloned again, ran `./install.sh` with the following result:

```
# ./openplc_editor.sh
Traceback (most recent call last):
  File "/home/user/Downloads/OpenPLC_Editor/./editor/Beremiz.py", line 32, in <module>
    import wx
  File "/home/user/Downloads/OpenPLC_Editor/.venv/lib/python3.9/site-packages/wx/__init__.py", line 17, in <module>
    from wx.core import *
  File "/home/user/Downloads/OpenPLC_Editor/.venv/lib/python3.9/site-packages/wx/core.py", line 12, in <module>
    from ._core import *
ImportError: libjpeg.so.9: cannot open shared object file: No such file or directory
```

thiagoralves, 2024-01-26:
I don’t know where this libjpeg.so.9 is coming from. I tested on other Ubuntu based systems and it worked just fine. I’m wondering if the issue is related to a precompiled wxPython who for your system. Try to force pip to recompile wxPython instead of using a precompiled binary with --no-cache-dir --no-binary :all:

closed 2024-03-22 by thiagoralves