thiagoralves/OpenPLC_Editor #100: Error finding libraries after they were re-downloaded for P1AM

njpatter, 2024-02-01:
When compiling for a P1AM, the .c/.h files were re-downloaded even though they had previously been installed.  Afterwards, every time it tries to compile the following errors are in the console window when transferring to the PLC:

`
[90mC:\Users\user_name_here\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.14[0m
loading library from C:\Users\user_name_here\OpenPLC_Editor\editor\arduino\__pycache__: invalid library: no header files foundloading library from C:\Users\user_name_here\OpenPLC_Editor\editor\arduino\bin: invalid library: no header files foundloading library from C:\Users\user_name_here\OpenPLC_Editor\editor\arduino\examples: invalid library: no header files foundloading library from C:\Users\user_name_here\OneDrive\Documents\Arduino\libraries\DarkArduinoTheme-master: invalid library: no header files foundloading library from C:\Users\user_name_here\OneDrive\Documents\Arduino\libraries\experimental: invalid library: no header files found
 
Uploading program to Arduino board at COM3...
Atmel SMART device 0x10010005 found
Device       : ATSAMD21G18A
Chip ID      : 10010005
Version      : v2.0 [Arduino:XYZ] Jun 25 2019 14:33:08
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
Boot Flash   : true
BOD          : true
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
done in 0.841 seconds
 
Write 18608 bytes to flash (291 pages)
 
[======                        ] 21% (64/291 pages)
[=============                 ] 43% (128/291 pages)
[===================           ] 65% (192/291 pages)
[==========================    ] 87% (256/291 pages)
[==============================] 100% (291/291 pages)
done in 0.148 seconds
 
Verify 18608 bytes of flash with checksum.
Verify successful
done in 0.016 seconds
CPU reset.
New upload port: COM3 (serial)
 
Done!
`

It seems like it is completing successfully, but then the PLC does not have operating code

njpatter, 2024-02-02:
I teach a course on automation and this is happening on 4-5 student computers.  As far as I know the remaining 20-25 are not seeing this problem.  I compared one of their compilation/transfer outputs to what I see on my computer, which is able to compile/transfer without issue.  The following are the differences that I can find:

1. FQBN statement missing from mine, but present in theirs:
`FQBN: arduino:samd:mkrzero `
2. All of my command line arguments, I'm guessing subprocess calls, are encapsulated in quotes with double \\, whereas theirs are not:
Mine:
`Detecting libraries used...
"C:\\Users\\... etc etc etc" `
Theirs
`Detecting libraries used...
C:\Users\... etc etc etc`

3. My temp folders are labeled as 'arduino-sketch-randomString' where theirs are nested folders 'arduino\sketches\randomString'

4. They see the output I noted above and mine just successfully uploads.

thiagoralves, 2024-02-02:
Start by checking if all of you are running on the same version of the editor. Most likely it is not. Is the problem happening on the most recent version or on an older version?

thiagoralves, 2024-02-26:
Awesome