thiagoralves/OpenPLC_Editor #102: Project Forked?

TheColonel2688, 2024-02-16:
The website looks to be about [half rebranded](https://autonomylogic.com/). And there are references to a different editor who's repo can be found here. https://github.com/beremiz/beremiz

I am very confused as to what is going on. I don't see any announcements or blog posts.

thiagoralves, 2024-02-16:
Hi TheColonel2688 !

The OpenPLC project is still the same, with the same name and logo. Autonomy is the company I created last year to handle OpenPLC since it has grown significantly in recent years. That's why the domain has changed and all requests to openplcproject.com are now being redirected to autonomylogic.com.

Beremiz is the core of the OpenPLC Project, in the same way that Wiring is the core of Arduino. This is stated in the Autonomy website:
[image]

I did write a short statement about this change in the forum: https://openplc.discussion.community/post/autonomy-12746686

Maybe I should write some more about this in an official announcement, but after almost a year, only three people (you included) asked this question, so I don't think there is a lot of interest in this anyway.

closed 2024-02-16 by thiagoralves

TheColonel2688, 2024-02-16:
Ok, but what is the difference between this repo we are in, and this repo https://github.com/beremiz/beremiz?

thiagoralves, 2024-02-17:
OpenPLC is focused on open source platforms like Arduino and Raspberry Pi boards. It is also packaged to “just work” on Windows, Linux and macOS. That’s why it has only a subset of Beremiz features that have been tested to work on all platforms. Beremiz has more features and is more generic. They also provide their own runtime that can run on Windows and Linux targets.