thiagoralves/OpenPLC_Editor #108: ST creation problem

fehimiltek, 2024-03-22:
hello I just realised that when i connect my output to a parameter on st file its appears to top of program let me show 

[image]
[image]

I think "Q1MODE := CNC_Mode0.Mode_Num;" line should be at the end

regards

thiagoralves, 2024-03-22:
It is hard to make sense of these screenshots. They are cropped, only small portions of the code and diagram are shown.

fehimiltek, 2024-03-23:
let me explain with another clear example
[image]
[image]

i think it should be like

`PROGRAM program0
  VAR
    adding0 : adding;
    LocalVar0 : BOOL;
    LocalVar1 : BOOL;
  END_VAR
  VAR
    LocalVar2 AT %MB1000 : BYTE;
  END_VAR

  LocalVar1 := LocalVar0;
  adding0(EN := LocalVar1, LocalVar0 := 10, LocalVar1 := 10);
  IF adding0.ENO THEN
      LocalVar2 := adding0.LocalVar2;
  END_IF;
END_PROGRAM`

regards