thiagoralves/OpenPLC_Editor #110: Issue installing PLC Editor on Debian 12

bionicsgeek, 2024-04-07:
Hi!

I encountered a problem running the PLC Editor software after installation. 
When I keyed in the command: ./openplc_editor.sh
It prompts the error: ./openplc_editor.sh: line10:  ./ .venv/bin.python3: No such file or directory.

How do i solve this issue?
thanks!

thiagoralves, 2024-04-07:
It seems that something went wrong during the installation of dependencies, most likely Python.

psanio, 2024-04-19:
Hi  guys,

I recently installed OpenPLC_Editor in two Debian based distributions. 

In LinuxMint 21.3 Virginia and MX Linux 23.2 Libretto:

- Changed the way to add DeadSnake repos following this a tutorial in DigitalOcean (ref1) extracting the data (key/server) from Launchpad (ref2):
``` Shell
sudo gpg --homedir /tmp --no-default-keyring --keyring /usr/share/keyrings/R.gpg --keyserver keyserver.ubuntu.com --recv-keys F23C5A6CF475977595C89F51BA6932366A755776
```
``` Shell
echo "deb [arch=amd64 signed-by=/usr/share/keyrings/R.gpg] https://ppa.launchpadcontent.net/deadsnakes/ppa/ubuntu jammy main" | sudo tee /etc/apt/sources.list.d/deadsnake.list > /dev/null
```
Additionaly in MX Linux 23.2 Libretto:
- added an additional package:
``` Shell
sudo apt install libgtk-3-dev
```
Regards,
Pablo.

ref1:
https://www.digitalocean.com/community/tutorials/how-to-handle-apt-key-and-add-apt-repository-deprecation-using-gpg-to-add-external-repositories-on-ubuntu-22-04#addendum-adding-an-external-repository-using-a-keyserver
ref2:
https://launchpad.net/~deadsnakes/+archive/ubuntu/ppa

JasperGraver, 2024-09-12:
i seem to have the same problem is this solved already if yes then how?

psanio, 2024-09-12:
> i seem to have the same problem is this solved already if yes then how?

Which distro and version are you using?

As far as I see the install.sh script has the original way of adding the deadsnake repo that in some distros dont work.

You car try replacing the add-apt-repository line with my suggetsion above

Regards,
Pablo.

JasperGraver, 2024-09-13:
I am on a project trying to do a few things with openplc on a raspberry pi 4b and i dont know alot about it and trying to install it via the way the videos/the README map doesnt work 

Im using a:
Raspberry Pi 4b
on it is Raspberry Pi OS (Rasbian) 64 bit 

I will look into your solution and i will try to do it

Kind Regards,
Jasper.

JasperGraver, 2024-09-13:
If you have any tips for me they are welcome !!! :)

JasperGraver, 2024-09-13:
I tried your thing and it didnt really work but as i told earlier i dont really know alot about these things so i could have done it wrong

thiagoralves, 2024-09-13:
The editor won't run on a Raspberry Pi. You need to run this on a x86 PC.

JasperGraver, 2024-09-13:
> The editor won't run on a Raspberry Pi. You need to run this on a x86 PC.

Can I run the editor on a separate pc and then run the runtime on my raspberry pi so I can use it raspberry pi to run it but write my things on a separate pc? 

Bc I was looking on the internet for it and it says a lot that I can be done 

kind regards,

jasper graver

thiagoralves, 2024-09-13:
Yup, that’s exactly how it was intended to be used. Runtime on a PLC device (raspberry pi in your case), Editor on your desktop/laptop.

JasperGraver, 2024-09-13:
Okay thanks but does that fix the .venv problem?

thiagoralves, 2024-09-13:
Which .venv problem? The runtime is another piece of software separated from the editor.

JasperGraver, 2024-09-13:
If I download the editor on the pc I have will it work like it’s used to work or will the problem that we are typing under still occur?

thiagoralves, 2024-09-13:
It should work just fine

JasperGraver, 2024-09-13:
Okay thanks I’ll check Monday and have a nice night love the openplc content ❤️

thiagoralves, 2024-09-13:
Is you PC running Linux, Windows or macOS? On Linux your mileage may vary depending on the distro you’re using…

JasperGraver, 2024-09-13:
Im running my own pc on windows

JasperGraver, 2024-09-17:
@thiagoralves 

I get the issue :

Couldn't find ./python/.venv/Scripts/python.exe

Can u help me?

JasperGraver, 2024-09-17:
pythonw.exe

typo
:D

thiagoralves, 2024-09-17:
Did you move the editor folder to another place after installing it? Try deleting the .venv folder and open the editor again. You must be connected to the Internet on that first launch as it will try to download and install dependencies.

JasperGraver, 2024-09-17:
@thiagoralves 

it says it can't find the path to setup the virtual environment 
and it also says: it can't find the path to install the dependencies

JasperGraver, 2024-09-17:
i didnt move it to another place and i cant seemt to find the .venv folder and i am connected to the internet

thiagoralves, 2024-09-17:
The startup script checks if the .venv folder exists inside C:\Users\[your-username]\OpenPLC_Editor\python\.venv. If that folder exists, then it tries to run the editor using the .venv python. If the folder doesn't exist, then it starts the .venv installation which includes installing all required libraries and dependencies. Perhaps something happened during the installation of the editor that didn't create this .venv properly. If you delete the .venv folder you will force the startup script to recreate it. I'm trying to find a way to make the editor more portable and don't depend on this initial .venv setup. If I manage to make it work I will let you know. It will make the installation process more smooth on Windows.

JasperGraver, 2024-09-20:
@thiagoralves 
i uninstalled it and re installed it fully twice and now it works for some reason :O

Thanks for your help <3

Keep up the good work

Kind regards,

Jasper Graver

thiagoralves, 2024-09-20:
Oh, I forgot to mention it to you, but I actually fixed the problem on the latest version of the installer: https://openplc.discussion.community/post/portable-openplc-editor-13493205?trail=15#1

If you downloaded the installer again from the Autonomy website, then probably that’s why it is working now.

I started this bug fix task because of your report and once it was finished I forgot to report back to you… 🤦🏻‍♂️ sorry!

JasperGraver, 2024-09-20:
Its okay :D im happy you fixed it and very grateful for your help <3

i was looking into it at home with my dad and i was like how does it work here but not on my work laptop but it was because you fixed it XD

Have a nice day! 
And thank you again!!!