thiagoralves/OpenPLC_Editor #117: Crash when adding power rail

jputcu, 2024-06-23:
I cloned the editor, ran it on Manjaro, tried to make a power rail but got a crash.
Python 3.12.3 

~~~~
12:31:52 PM: Debug: Adding duplicate image handler for 'Windows bitmap file'
12:31:52 PM: Debug: Adding duplicate animation handler for '1' type
12:31:52 PM: Debug: Adding duplicate animation handler for '2' type
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/wx/core.py", line 3427, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jorisp/Workspace/3rdparty/OpenPLC_Editor/editor/editors/Viewer.py", line 2706, in AddNewPowerRail
    powerrail = LD_PowerRail(self, values["type"], id, values["pin_number"])
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jorisp/Workspace/3rdparty/OpenPLC_Editor/editor/graphics/LD_Objects.py", line 53, in __init__
    self.SetType(type, connectors)
  File "/home/jorisp/Workspace/3rdparty/OpenPLC_Editor/editor/graphics/LD_Objects.py", line 240, in SetType
    self.AddConnector()
  File "/home/jorisp/Workspace/3rdparty/OpenPLC_Editor/editor/graphics/LD_Objects.py", line 134, in AddConnector
    self.InsertConnector(len(self.Connectors))
  File "/home/jorisp/Workspace/3rdparty/OpenPLC_Editor/editor/graphics/LD_Objects.py", line 144, in InsertConnector
    self.RefreshConnectors()
  File "/home/jorisp/Workspace/3rdparty/OpenPLC_Editor/editor/graphics/LD_Objects.py", line 196, in RefreshConnectors
    connector.SetPosition(wx.Point(self.Size[0], position))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Point(): arguments did not match any overloaded call:
  overload 1: too many arguments
  overload 2: argument 2 has unexpected type 'float'
  overload 3: argument 1 has unexpected type 'int'
  overload 4: argument 1 has unexpected type 'int'
~~~~

Fixed by adding extra `int` cast:
~~~~
editor/graphics/LD_Objects.py:194  position = int(round((self.Pos.y + position) / scaling[1]) * scaling[1] - self.Pos.y)
~~~~

jputcu, 2024-06-23:
Same for:
~~~~
editor/graphics/LD_Objects.py:868 position = int(round((self.Pos.y + position) / scaling[1]) * scaling[1] - self.Pos.y)
~~~~

thiagoralves, 2024-06-23:
You need Python 3.9 and wxPython 4.2.0

closed 2024-06-23 by thiagoralves