thiagoralves/OpenPLC_Editor #133: Bug :((

JasperGraver, 2024-10-07:
I made a little setup in OpenPLC Editor (the one in the yt vid) to learn a bit abt it and i clicked on start PLC Sim and then ran it and it ran fine then stopped the Sim and then tried starting it again and it gives me a bug report error thing i cant seem to find the bug report (im working on a laptop with limited privs so that may be a reason

maby someone else can try to remake the error

so in short
1. i made a little setup from the yt vid
2. started the PLC simulator
3. Clicked the debugging tool
4. stopped the PLC simulator
5. Tried starting the simulator again
6. error

[image]

JasperGraver, 2024-10-07:
And yes i tried to search C:/Users/........ 
It says i dont have perms

JasperGraver, 2024-10-07:
In the console Log it says cleaning buildmap but it doesnt go any further then saying it :/

JasperGraver, 2024-10-10:
@thiagoralves you have a solution for this?

thiagoralves, 2024-10-10:
I can’t reproduce this. Make sure your antivirus or any other protection software is not blocking the debugger

JasperGraver, 2024-10-11:
Temp fix = deleting the build map of the program everytime 
nothing is blocking the debugger it just cant delete/clean the buildmap

ill look into the blockers

JasperGraver, 2024-10-11:
i didnt find anything else that would be able to block it from working ill just redownload the editor and i hope that fixes it and if it doesnt then ill just delete the build file everytime i wanna redo it

thiagoralves, 2024-10-11:
The editor does delete the build folder every time you do a build, and that's most likely the action that is being blocked, perhaps by your Windows Defender, which is MS native antivirus on Windows. Try temporarily disabling Windows Defender during your compilation and simulation just to verify if the problem goes away. If it does, then certainly that's the culprit.

JasperGraver, 2024-10-14:
Reinstalling seems to have fixed the problem thx for the help <3

closed 2024-10-14 by JasperGraver