thiagoralves/OpenPLC_Editor #144: Compilation failed for missing globals (ESP [3.3.0])

DarioDaF, 2025-08-02:
I tried compiling the internal provided example for some ESP32 boards on Windows 11 (nothing arduino related installed before running the editor)
all the install and downloads process worked fine, the same sketch compiles with no error on Arduino Uno, using `ESP32 Generic [3.3.0]` that was automatically downloaded (same for WROOM, S3 and so on...) errors on compilation on the `src` part (dropping most of the output for clarity):

```
"C:\\Users\\REDACTED_USER\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp-x32\\2411/bin/xtensa-esp32-elf-gcc" "-Ieditor\\arduino\\src\\lib" -MMD -c -MMD -c "@C:\\Users\\REDACTED_USER\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32/flags/c_flags" -w -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32:UploadSpeed=230400\"" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 "@C:\\Users\\REDACTED_USER\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32/flags/defines" "-IC:\\Users\\REDACTED_USER\\OpenPLC_Editor\\editor\\arduino\\examples\\Baremetal" -iprefix "C:\\Users\\REDACTED_USER\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32/include/" "@C:\\Users\\REDACTED_USER\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32/flags/includes" "-IC:\\Users\\REDACTED_USER\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.5-b66b5448-v1\\esp32/qio_qspi/include" "-IC:\\Users\\REDACTED_USER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\cores\\esp32" "-IC:\\Users\\REDACTED_USER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\variants\\esp32" "-IC:\\Users\\REDACTED_USER\\OpenPLC_Editor\\editor\\arduino\\src" "-IC:\\Users\\REDACTED_USER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\libraries\\WiFi\\src" "-IC:\\Users\\REDACTED_USER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\libraries\\Network\\src" "@C:\\Users\\REDACTED_USER\\AppData\\Local\\Temp\\arduino\\sketches\\F5180C918FD3EC4B5BC148314ABDA258/build_opt.h" "@C:\\Users\\REDACTED_USER\\AppData\\Local\\Temp\\arduino\\sketches\\F5180C918FD3EC4B5BC148314ABDA258/file_opts" "C:\\Users\\REDACTED_USER\\OpenPLC_Editor\\editor\\arduino\\src\\POUS.c" -o "C:\\Users\\REDACTED_USER\\AppData\\Local\\Temp\\arduino\\sketches\\F5180C918FD3EC4B5BC148314ABDA258\\libraries\\src\\POUS.c.o"

[...]

In file included from C:\Users\REDACTED_USER\OpenPLC_Editor\editor\arduino\src\POUS.h:4,
                 from C:\Users\REDACTED_USER\OpenPLC_Editor\editor\arduino\src\POUS.c:1:
C:\Users\REDACTED_USER\OpenPLC_Editor\editor\arduino\src\POUS.c: In function 'ARDUINOSEG_init__':
editor\arduino\src\lib/accessor.h:71:30: error: implicit declaration of function '__GET_GLOBAL_CURRENT_COUNT' [-Wimplicit-function-declaration]
   71 |                 name.value = __GET_GLOBAL_##global();\
      |                              ^~~~~~~~~~~~~
C:\Users\REDACTED_USER\OpenPLC_Editor\editor\arduino\src\POUS.c:7:3: note: in expansion of macro '__INIT_EXTERNAL'
    7 |   __INIT_EXTERNAL(INT,CURRENT_COUNT,data__->CURRENT_COUNT,retain)
      |   ^~~~~~~~~~~~~~~
editor\arduino\src\lib/accessor.h:71:28: error: assignment to 'IEC_INT *' {aka 'short int *'} from 'int' makes pointer from integer without a cast [-Wint-conversion]
   71 |                 name.value = __GET_GLOBAL_##global();\
      |                            ^
C:\Users\REDACTED_USER\OpenPLC_Editor\editor\arduino\src\POUS.c:7:3: note: in expansion of macro '__INIT_EXTERNAL'
    7 |   __INIT_EXTERNAL(INT,CURRENT_COUNT,data__->CURRENT_COUNT,retain)
      |   ^~~~~~~~~~~~~~~
Multiple libraries were found for "Arduino_OpenPLC.h"
  Used: C:\Users\REDACTED_USER\OpenPLC_Editor\editor\arduino\src
  Not used: C:\Users\REDACTED_USER\OpenPLC_Editor\editor\arduino\src_Good
Multiple libraries were found for "WiFi.h"
  Used: C:\Users\REDACTED_USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0\libraries\WiFi
  Not used: C:\Users\REDACTED_USER\Documents\Arduino\libraries\WiFiNINA
Using library src in folder: C:\Users\REDACTED_USER\OpenPLC_Editor\editor\arduino\src (legacy)
Using library WiFi at version 3.3.0 in folder: C:\Users\REDACTED_USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0\libraries\WiFi 
Using library Networking at version 3.3.0 in folder: C:\Users\REDACTED_USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0\libraries\Network 

Used library Version Path
src                  C:\Users\REDACTED_USER\OpenPLC_Editor\editor\arduino\src
WiFi         3.3.0   C:\Users\REDACTED_USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0\libraries\WiFi
Networking   3.3.0   C:\Users\REDACTED_USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0\libraries\Network

Used platform Version Path
esp32:esp32   3.3.0   C:\Users\REDACTED_USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0
Error during build: exit status 1
$? = 1
```

This gets fixed normally by simply providing `Config0.h` include in the POUS header file

The build with Arduino Uno is exactly the same, POUS generated file do not seem to contain Config0 (I am not aware on how it's injected into the make process but it seems like avr gcc does a bit more than just "compiling" with the provided flags since it doesn't have any command line flags that pop out on the same compile line that fails here)

thiagoralves, 2025-08-02:
Update your editor to the latest version and it will work

closed 2025-08-02 by thiagoralves

DarioDaF, 2025-08-02:
Thanks can confirm it works perfectly, I didn't expect to be any update because I just run like 10 min ago the freshly downloaded official installer from `https://autonomylogic.com/`